pybpod/pybpod #88: Update Questions link in docs to point from bitbucket to this git-hub repo

oraby, 2019-10-29:
The [Questions](https://pybpod.readthedocs.io/en/v1.7.8/about/about.html) section in readthedocs page refer to the bitbucket project, rather than this github repo.

MicBoucinha, 2019-10-29:
Thanks, I missed that link. I will fix it.
Please let me know if you have problems regarding using PyBpod with Bpod v0.5 as you mentioned in the bitbucket repository.

MicBoucinha, 2019-10-29:
Fixed in the develop branch, will be available on the next release. Thanks for the report, I missed some other links in other parts of the documentation too.

closed 2019-10-29 by MicBoucinha

oraby, 2019-10-30:
> Please let me know if you have problems regarding using PyBpod with Bpod v0.5 as you mentioned in the bitbucket repository.

I will try it later this week and report back regardless of the result.
Thanks for the support 👍