pybpod/pybpod #89: ModuleNotFoundError: No module named 'sca'

Pentrimex333, 2020-03-03:
I am attempting to import Bpod from pybpodapi.protocol, but it keeps running into the above mentioned error. To note I have pip installed both pybpod and pybpodapi. The specific line it references as causing the error is line 12 from session.py that states:

from sca.formats import csv

I am not sure how to reproduce this error, other than to install the pybpod-api library on a naive PC and see if it happens.

This is the error in full:

ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-8-060793c58f25> in <module>
----> 1 from pybpodapi.protocol import Bpod, StateMachine

~\Anaconda3\lib\site-packages\pybpodapi\protocol\__init__.py in <module>
----> 1 from pybpodapi.bpod import Bpod
      2 from pybpodapi.state_machine import StateMachine
      3 from pybpodapi.bpod.hardware.output_channels import OutputChannel

~\Anaconda3\lib\site-packages\pybpodapi\bpod\__init__.py in <module>
----> 1 from pybpodapi.bpod.bpod_io import BpodIO
      2 
      3 
      4 class Bpod(BpodIO):
      5     pass

~\Anaconda3\lib\site-packages\pybpodapi\bpod\bpod_io.py in <module>
      6 
      7 from confapp import conf as settings
----> 8 from pybpodapi.bpod.bpod_com_protocol_modules import BpodCOMProtocolModules
      9 from pybpodapi.com.messaging.session_info import SessionInfo
     10 

~\Anaconda3\lib\site-packages\pybpodapi\bpod\bpod_com_protocol_modules.py in <module>
      8 from pybpodapi.com.arcom import ArduinoTypes
      9 from pybpodapi.bpod_modules.bpod_modules import BpodModules
---> 10 from pybpodapi.bpod.bpod_com_protocol import BpodCOMProtocol
     11 from pybpodapi.com.protocol.send_msg_headers import SendMessageHeader
     12 from pybpodapi.com.protocol.recv_msg_headers import ReceiveMessageHeader

~\Anaconda3\lib\site-packages\pybpodapi\bpod\bpod_com_protocol.py in <module>
      5 
      6 from confapp import conf as settings
----> 7 from pybpodapi.bpod.bpod_base import BpodBase
      8 from pybpodapi.bpod.hardware.channels import ChannelType
      9 from pybpodapi.bpod_modules.bpod_module import BpodModule

~\Anaconda3\lib\site-packages\pybpodapi\bpod\bpod_base.py in <module>
     28 from pybpodapi.com.messaging.state_transition import StateTransition
     29 
---> 30 from pybpodapi.session import Session
     31 
     32 from .non_blockingstreamreader import NonBlockingStreamReader

~\Anaconda3\lib\site-packages\pybpodapi\session.py in <module>
     10 from pybpodapi.com.messaging.state_occurrence import StateOccurrence
     11 
---> 12 from sca.formats import csv
     13 
     14 import sys

ModuleNotFoundError: No module named 'sca'

**Desktop (please complete the following information):**
 - OS: Windows 10 1909
 - PyBpod version 1.8.2

**Additional context**
Python 3.7 has been installed on this computer through Anaconda and Jupyter Notebook.

MicBoucinha, 2020-03-03:
Hi,

Can you check if you have the "safe-and-collaborative-architecture" package installed in your environment please?

Pentrimex333, 2020-03-03:
C:\Users\Felsenlab>pip install safe-and-collaborative-architecture
Requirement already satisfied: safe-and-collaborative-architecture in c:\users\felsenlab\appdata\local\programs\python\python37-32\lib\site-packages (0.2)
Requirement already satisfied: send2trash in c:\users\felsenlab\appdata\local\programs\python\python37-32\lib\site-packages (from safe-and-collaborative-architecture) (1.5.0)

Yep, this is what I got back from command prompt, so it appears that it has been installed.

MicBoucinha, 2020-03-04:
We didn't consider that use case (running in a Jupyter Notebook) since most of our users don't use them (mostly due to some of the limitation that they have).

However, if you want to use it in Jupyter Notebooks, don't forget that Jupyter kernels are disconnected from the shell that Jupyter uses. Since the packages used by the Jupyter Notebooks kernel aren't necessarily the same that are installed in the Python's environment, this might be your problem.

You can try to install it by using something along the lines of (**note: not tested**) inside your Jupyter Notebook:
```python
import sys
!{sys.executable} -m pip install pybpod-api
```

From your info, it seems that you are running everything globally. Can you try to create a new, clean conda environment, enter it and install from there to test if you still have the same issue?