pybpod/pybpod #90: Rotary Encoder Documentation

maxhoheiser, 2020-05-01:
Hi, I am currently trying to reverse engineer your scripts to implement my own tasks, but I am stuck with the rotary encoder.
I could not find any documentation on the wiki, do you have some additional documents?

Best regards Max

MicBoucinha, 2020-05-04:
Hi, 

you can try to look at the Rotary Encoder module documentation here:  https://pybpod.readthedocs.io/projects/pybpod-gui-plugin-rotaryencoder/en/latest/ 

It has a couple of examples on how to access the Rotary encoder by code. 

Let me know if that helps.

Thanks

maxhoheiser, 2020-05-06:
Hi Luis,
thanks for your input, I could move a bit further. I reverse engineered your trials scripts from the iblrig github.