pybpod/pybpod #91: 'command not found: start-pybpod' after installing pybpod (for developers)

ckarageorgkaneen, 2020-06-22:
I succesfully ran steps 1-6 from [Installation for developers](https://pybpod.readthedocs.io/en/v1.8.1/getting-started/install.html#installation-for-developers). However, running
```bash
start-pybpod
```
yields:
```bash
start-pybpod: command not found
```
No start-pybpod file or alias seems to be created... 

Am I missing something?

**To Reproduce**
Steps to reproduce the behavior:
1. Follow steps 1-6 from [Installation for developers](https://pybpod.readthedocs.io/en/v1.8.1/getting-started/install.html#installation-for-developers)

**Expected behavior**
What I expected is for a start-pybpod file or alias to be created and the PyBpod application to run.

**Desktop:**
 - OS: Ubuntu 20.04
 - PyBpod version: 1.8.2

MicBoucinha, 2020-06-22:
Hi,

The start-pybpod command is installed when installing the pybpod module and the install.py script is not doing that at the moment (I will issue a fix as soon as possible).

As such, within your virtual environment, you should go to the `base/pybpod` folder and install it through
`pip install -e .`

After that, the `start-pybpod` command will be available and working properly.

Thanks for finding this issue (most people only install it to use the software).

All the best

ckarageorgkaneen, 2020-06-22:
It worked, thanks.

MicBoucinha, 2020-06-22:
Great!

ckarageorgkaneen, 2023-07-04:
> Hi,
>
> The start-pybpod command is installed when installing the pybpod module and the install.py script is not doing that at the moment (I will issue a fix as soon as possible).
>
> As such, within your [...]

**Note**: Doing as @MicBoucinha points out does indeed make the `start-pybpod` available. However, for developers I would not recommend doing that since it also installs fixed versions which might overwrite individual packages (e.g. changed ones that developers might be wanting to test out) that were previously installed while doing `python utils/install.py`, which defeats the purpose of development. Meaning, why install `pybpod` the developer way if we're just going to install it the user way afterwards?

So, in order to get `start-pybpod` to become available, instead of going into `base/pybpod` and doing `pip install -e.`, I recommend starting the GUI the following way:

```bash
python -c 'import pybpodgui_plugin.__main__ as Main; Main.start()'
```

See https://github.com/HenryJFlynn/mouse2afc/commit/2c4b4cc915c0dcfb15ec64adf210d7286c2124b6.

closed 2023-07-04 by ckarageorgkaneen