pybpod/pybpod #92: Updating protocol/experiment settings mid-session?

Pentrimex333, 2020-09-14:
Currently I have been unable to find a method by which the settings in a protocol can be adjusted during a session; for example, say I am running a mouse on a mix2afc experiment where they choose between a left/right port, but during a session I am noticing that the mouse is exhibiting a left-side bias. The way the protocol is currently handled by Pybpod, I cannot access the list that determines the correct side per trial within my protocol, and thus am forced to stop the current session and implement a new protocol (let's call it "L-side bias protocol") and try to combine the data from both sessions for the day outside of Pybpod.

I would like for there to be a plugin or GUI update that allows me to access these protocol trial settings (correct choice per trial, delay times, etc.) so that if I need to adjust them in the middle of a session, I would be able to do so without completely stopping and starting a new protocol/session each time.

If this is too problematic to enact without specific protocol writing rules to implement these mid-session updates, then another possibility would be a feature that allows me to combine data from two sessions within the GUI into a single csv file; you could create a new right-click feature that offers a "Combine with" or "Attach to" function where the chosen session folder has its basic info at the top (filename, mouse name, basic info stuff at the top) removed and the rest attached to a pre-existing session csv file.

Thanks guys, let me know if you need a better description or any other info.

MicBoucinha, 2020-10-07:
Hi,

How are you keeping that list? Can you show me some code so I can try to help?

HenryJFlynn, 2023-08-02:
Hello, I would also like a feature like this and am wondering if there as been any progress?