pybpod/pybpod #95: Is Pybpod compatible with BPod Generation 1?

alisonsin, 2020-11-23:
Hello, when I have used the same protocol that I have used for Bpodr2 on the BPod generation 1 instead, I could not run it. Here is the error message I got from Terminal. I would like to ask how should I modify the code in order for it to also work for BPod generation 1? 

Thank you very much!
[image]

MicBoucinha, 2020-11-24:
Hi,

PyBpod is mostly dependent on the Bpod's firmware version that you have installed in your device. A while back I tested with both a Bpod v0.7 and a Bpod v2 running simultaneously and PyBpod seemed to handle things properly (both had firmware v22, which are available on the official Sanworks repository here: https://github.com/sanworks/Bpod_StateMachine_Firmware/tree/v22/Preconfigured)

I will try to run a few tests at the end of the week when I have access to the devices.

alisonsin, 2020-11-24:
Ok, Thank you! I believe it has to do with the init files in the hardware
module, in which  whether script bpod0_7_9_fw9//13/20 is read.

alisonsin, 2020-11-24:
Is there a way I could get rid of this error:
 hardware.inputs_enabled[i] = settings.BPOD_BEHAVIOR_PORTS_ENABLED[j]
IndexError: list index out of range

I am currently using BPod r1. which I believe uses firmwareversion = 20. Thank you!

MicBoucinha, 2020-11-27:
Hi,

Can you try to add the config option `TARGET_BPOD_FIRMWARE_VERSION = "20"` to your user_settings.py file please and try again?

The default version used in PyBpod is version "22", which was released over 2 years ago.

If there is nothing preventing you from upgrading your device's firmware, I would recommend it.

alisonsin, 2020-12-01:
Hi, I have done so. But I got an error message saying that 'future firmware
detected. Please update the Bpod python software. How should I proceed?
Thank you!

MicBoucinha, 2020-12-03:
Hi,

If you are getting that message after changing `TARGET_BPOD_FIRMWARE_VERSION = "20"` it means that your firmware if higher than version 20. If you have firmware with version 22, the issue that you are seeing might be different. 

I will try to find a Bpod v1 to further understand the issue.

alisonsin, 2020-12-03:
Thank you very much!!