pybpod/pybpod #97: globaltriggercancel event seems not to work

rozmar, 2020-12-01:
start a looped global timer, enable then try to disable it. 
For me, it doesn't work, the global timer just goes on and on.

set up global timer:

    sma.set_global_timer(timer_id=1, 
        timer_duration=.1, 
        on_set_delay=0, 
        loop_mode=1,
        send_events=1)

start the global timer in a state with ('GlobalTimerTrig', 1)
try to stop the global timer in the next state with ('GlobalTimerCancel', 1) 

**Expected behavior**
The global trigger (or at least its output) should stop when the output action of a state is ('GlobalTimerCancel', 1) - given 1 is the global trigger index

**Desktop (please complete the following information):**
Win10
PyBpod version 1.8.2

hanhou, 2021-04-14:
Same here: `(GlobalTimerCancel, 1)` won't stop the global timer #1 in pybpod (1.8.2).

A related question: can pybpod START (or STOP) multiple global timers at the same time? In the [Matlab version of bpod](https://sites.google.com/site/bpoddocumentation/bpod-user-guide/function-reference-beta/setglobaltimer_beta), we can use binary code such as `{'GlobalTimerTrig', '1111'}` and  `{'GlobalTimerCancel', '1111'}`. In pybpod, the second argument must be an integer (I tried string and got an error), so I expected that pybpod would convert this decimal to binary and switch on/off the corresponding timers. However, in pybpod, `('GlobalTimerTrig', 3)` only triggers timer #3, rather than both timers #1 and #2 (3 --> '11'). And of course, `('GlobalTimerCancel', 3)` doesn't have any effect, as indicated by the original question.

hanhou, 2021-04-21:
Hi Marton,

I dug into the source code of pybpod-api and found the fix! Please see the above pull request.

Now not only does the `GlobalTimerCancel` work properly, but you can also control more than one global timers at the same time. Here's an example:

[image]

With your already installed pybpod, a quick fix would be to simply change the code in `$YourPythonEnvironment$\Lib\site-packages\pybpodapi\state_machine\state_machine_base.py`.

Please let me know whether this also works for you. Thanks!

Han