pybpod/pybpod #98: TTL output during whole session for level high camera acquisition

histun, 2020-12-02:
I'm using Bpod R2. Using a TTL output, I would like to send out a constant TTL for the whole session(ex, 1hr) to start/end video (Flir, blackfly) acquisition. 

Below is the task where I use two lists for looping trials with different valve time and ITI.
1) time_open : set valve duration
2) ITI_open : set ITI 

for (i,t) in zip(time_open,ITI_time):
	sma = StateMachine(my_bpod)
	#Loop between 'ITI' and 'Open_valve' state
	sma.add_state(
		state_name='ITI',
		state_timer=t,		
		state_change_conditions={Bpod.Events.Tup: 'open_valve'},										
		output_actions=[]) 
	sma.add_state(
		state_name='open_valve',
		state_timer=i, 	 #Open valve in randomly chosen duration	
		state_change_conditions={Bpod.Events.Tup: 'exit'},
		output_actions=[(Bpod.OutputChannels.Valve, 1)]) # open valve

	my_bpod.send_state_machine(sma)  # Send state machine description to Bpod device
	my_bpod.run_state_machine(sma)  # Run state machine

I have used manual overdrive to turn on BNC, but it turns off when the state machine starts.
Ideally, I would like to turn on the BNC before and turn off after the code above.

Thank you!

MicBoucinha, 2020-12-04:
Hi,

What you are suggesting isn't possible directly with Bpod's hardware because whenever you send a new StateMachine to the device, everything gets reset and there isn't a way to prevent that. That is the way the device works.

A suggestion would be to have a external device that you can control independently of your trials loop to send those start and stop TTL (an Arduino or maybe a second Bpod) that would be independent of that trial loop. I am not currently seeing an alternative. You can try to post a question on Sanworks' support forum to see how other people handle similar situations to yours.

histun, 2020-12-04:
Thanks. I asked Josh, too, and it seems not possible.
I decided to use the logic block features in blackfly cameras to start/finish frame acquisition with two TTL outputs before and after the above loop trials.

MicBoucinha, 2020-12-04:
I'm glad you found a solution.

closed 2020-12-04 by MicBoucinha