pybpod/pybpod #104: No Trigger to Pinch Valve

srhaiken, 2022-01-14:
Suddenly, the pinch valve stopped working on the behavior rig. I am confident the issue at hand is the computer not sending a trigger to the pinch valve to open and close, but the software does not provide any errors.

I am confident this is the issue because the debugging I have done thus far includes:
- Restarted computer 
- Unplugged and Replugged all ports 
- Changed to alternative pinch valves (tried a total of 3) 
- Changed controlling board 
- Reset BPOD 
- Changed BPOD 
- Changed ethernet cable 
- Plugged the valve into a power jack and plugged that into a power strip and made sure it is opening and closing properly (which it is) 
- Used a LED anode to see if it would light up when the pinch valve should be triggered, but it didn't 
- Used the whole reward system on another rig and was able to trigger the water 

Removing the IBLRig repository, deleting IBLEnv, re-cloning the repository from Github, and re-installing also did not fix the problem. Any thoughts or suggestions?

MicBoucinha, 2022-01-18:
Hi,

Can you reproduce the issue with a simple example without the IBLRig dependency to confirm that it is an issue with PyBpod itself?

Also, did the valve worked previously with the same version of PyBpod?

srhaiken, 2022-01-18:
Using an oscilloscope and wave form generator, I was able to reproduce the issue without the IBLrig dependency.

Yes, the valve worked previously with the same version of PyBpod.

MicBoucinha, 2022-01-18:
So, if it worked previously with the same version of PyBpod, how can this be a direct issue with PyBpod?

Can you confirm that you have the same package dependencies versions installed in the different computers you tested? It is the only thing that I am currently seeing that could be the cause of your issues.

srhaiken, 2022-01-18:
I don't know if it is a direct issue with PyBpod, I am simply asking for help and advice since I don't know what caused the rig to stop running nor do I know how to fix it. 

The computers are a clone of each other so I can confirm they are the same package dependency versions.

MicBoucinha, 2022-01-18:
If the package dependency version are the same, this might be an issue unrelated with PyBpod.

If you haven't yet, perhaps you can try directly on the IBLRig repository (https://github.com/int-brain-lab/iblrig) since they have experience with your exact setup.

srhaiken, 2022-01-18:
Thank you, will do.

closed 2022-01-18 by srhaiken